zeevy/grblcontroller #178: It won't let me use it

Danixxdd, 2024-05-13:
Download the application through APKPure but it won't let me use it. I connect to the HC-05 but I can't do anything. I need help, please. 
[image]
[image]

zeevy, 2024-05-13:
You can download the aab file from here https://github.com/zeevy/grblcontroller/releases

Looks like your firmware is not supported, application supports only grbl 1.1f

You can check your firmware or any other error messages in console tab

Danixxdd, 2024-05-13:
What can I do to solve this problem please help

zeevy, 2024-05-15:
This app works only with grbl 1.1f firmware, if you have old one then you have two options
1. carefully upgrade to 1.1f
2. leave the old device as is, buy new board and flash grbl 1.1f

More information of upgrading the firmware or flashing new will be available in grbl repo

closed 2024-05-15 by zeevy