zeevy/grblcontroller #180: Restore the operation of the Grbl Controller after a failure

seriogik777, 2025-07-22:
Hello. I am interested in whether it is possible to restore the operation of the Grbl Controller after a failure, for example, to start working from a specific line of g-code? I have a free version, maybe the paid version of Grbl Controller  has this feature or maybe it will be in future versions?

zeevy, 2025-08-19:
It has that feature in paid version ( availale in playstore )
https://zeevy.github.io/grblcontroller/file-streaming-panel.html (Streaming/Job Resume (Grbl Controller + version only)
)

seriogik777, 2025-08-21:
But the paid version says that resume work does not support incremental distance G-code 91. Are there other ways?

zeevy, 2025-08-22:
Resume only supports absolute coordinates